ARM-software/sbsa-acs #484: test_ras009.c provides insufficient error handling

ablackham52, 2024-09-27:
The exception handler provided in the test does not handle the containable error that was injected. There needs to be an appropriate way for the test to handle the containable error.

rajatgoyal47, 2024-10-09:
Hello,

As part of the exception handler in the test, we are updating the ELR to ensure that execution returns to the test as expected. 
Could you confirm if you've encountered any issues related to this? This would help us support you more effectively.

Regards,
ACS Team

rajatgoyal47, 2024-11-06:
Hello ablackham52,

Closing this issue for now. If you’re still experiencing the problem or have more details to share, please feel free to reopen this issue or create a new one.

Regards,
ACS Team

closed 2024-11-06 by rajatgoyal47