ARM-software/sbsa-acs #485: test_ras011.c needs to inject uncorrectable error

ablackham52, 2024-09-27:
The SBSA spec recommends that poison is generated when an uncorrectable error is detected. This test tries to inject a Correctable Error

rajatgoyal47, 2024-10-14:
Hi,

Thank you for highlighting this.

As RAS tests can vary due to their implementation-defined nature, they haven't been extensively verified on our end.

Could you please try changing the error type to ERR_UC and update the platform API to inject the error accordingly? You can refer to the following line for context:
https://github.com/ARM-software/sbsa-acs/blob/master/test_pool/ras/operating_system/test_ras011.c#L123

If further adjustments are needed to run or manage the errors, feel free to reach out, or alternatively, you're welcome to submit a pull request with your changes.

Regards,
ACS Team

rajatgoyal47, 2024-11-06:
Hello ablackham52,

Closing this issue for now. If you’re still experiencing the problem or have more details to share, please feel free to reopen this issue or create a new one.

Regards,
ACS Team

closed 2024-11-06 by rajatgoyal47