ARM-software/sbsa-acs #512: sbsa-acs still run PCIe tests even if there is no ECAM

sunnywang-arm, 2024-11-09:
sbsa still run PCIe test even if there is no ECAM

Two SBSA tests are still run even if there is no ECAM support, which is not consistent with bsa-acs
```
840 : PCIe RC,PE - Same Inr Shareable Domain
       PCI_IC_11
       Checkpoint --  1                           : Result:  SKIPPED
 601 : Check EA Capability                   
       No ECAMs discovered              
       S_L4PCI_2
       Checkpoint --  1                           : Result:  SKIPPED
```

The following code block is missing in val_sbsa_pcie_execute_tests()

  num_ecam = (uint32_t)val_pcie_get_info(PCIE_INFO_NUM_ECAM, 0);
  if (!num_ecam) {
      val_print(ACS_PRINT_WARN, "\n      *** No ECAM region found, Skipping PCIE tests ***\n", 0);
      return ACS_STATUS_SKIP;
  }

Sujana-M, 2024-11-11:
Hi @sunnywang-arm,

Thank you for raising this. 
For Test 601, We will make the change to check for ECAM much before other PCIe tests. 
Test 840 checks for a Root Complex in the system and RC can be present even if the ECAM is not present. So this test can be run even if ECAM are not present.
Please share your thoughts on Test 840. 

Thanks,
ACS Team

sunnywang-arm, 2024-11-11:
Thanks for promptly fixing the issue. @Sujana-M and @chetan-rathore.