ARM-software/sbsa-acs #513: sbsa-acs should show a failure rather than skipping test for No SMMU case

sunnywang-arm, 2024-11-10:
Now if the system doesn't have SMMU, sbsa-acs is showing "No SMMU Controller Found, Skipping SMMU tests..." rather than a failure. Per Table 2: SBSA level mapping summary in SBSA specification, it looks like a SMMU is required in SBSA level 3.  I'm not sure if we have a level 3 test to check the configurations mentioned in Table 2. If not, we will need to add a test for it. 

It looks like we should remove the code block below in val\sbsa\src\sbsa_execute_test.c.  
```
  num_smmu = val_iovirt_get_smmu_info(SMMU_NUM_CTRL, 0);
  if (num_smmu == 0) {
    val_print(ACS_PRINT_WARN, "\n     No SMMU Controller Found, Skipping SMMU tests...\n", 0);
    return ACS_STATUS_SKIP;
  }

```

chetan-rathore, 2024-11-11:
Hi @sunnywang-arm,

Agree, that the check is not required for SBSA and only needed in BSA.

On the SBSA L3 no smmu failure, currently as per SBSA L3 checklist required BSA SMMU rules are run.
Since, BSA doesn't mandates systems to have SMMU, smmu tests are skip if no SMMU.
We might need to add condition to the tests to fail, if run as part of SBSA L3.

Please share your thoughts.

Thanks,
Chetan

sunnywang-arm, 2024-11-11:
Thanks for the prompt follow-up, @chetan-rathore. 
As we discussed offline, Table 2 is indicative and not normative, which means SMMU is required in SBSA L4 rather than L3. Therefore, we can probably keep the code block for only L3.

chetan-rathore, 2024-11-12:
Hi @sunnywang-arm,

Currently there are no sbsa direct test at L3, and the code block was incorrectly skipping the SMMU L4+ tests if no smmu.
In future, when SBSA l3 test are added, we will make necessary changes to skip L3 test if no SMMU.

closed 2024-11-12 by chetan-rathore

sunnywang-arm, 2024-11-12:
Oh, I got it. I thought the fix was https://github.com/ARM-software/sbsa-acs/pull/514 (the one linked to this issue) but it turned out https://github.com/ARM-software/bsa-acs/pull/391 is the fix. I checked the wrong pull request.  Yeah, the fix looks good to me. Thanks, @chetan-rathore and @ajayswar-s 

https://github.com/ARM-software/bsa-acs/commit/20afb64ed385a5d0f86140c7c7646d3329c9e14f