ARM-software/sbsa-acs #521: Support for multiple PCIe segments

Azurithenkel, 2024-12-12:
Referred case from Vincent Gatine:

--------------

Hello,

I am using the UEFI sbsa test suite at https://github.com/ARM-software/sbsa-acs on my platform and I find that the test suite reports too many exercisers.

My platform has multiple PCIe controllers each within its own segment and only one of them has an actual exerciser. Correspondingly, the MCFG table has multiple entries corresponding to each segment/controller.

When the Exerciser tests run, the binary try to find all the exercisers in the system[1] by enumerating the BDFs found in the system[2] and matching them with an expected Vendor/Device Id[3].

But the BDF fetched from the built table are actually in the SBDF (Segment, Bus, Device, Function) Format and the `pal_is_bdf_exerciser()` function always takes
the first ECAM entry in the MCFG and adds it to the extracted Bus/Device/Function of the SBDF.

So given an exerciser on SBDF 0x4010000 and an unrelated device on SBDF 0x5010000, and given that the first entry in the MCFG is the segment 4.
The test suite reports two exercisers and tries to use both. Which does not work.

The function `pal_is_bdf_exerciser()` should probably be updated to handle multiple segments.

Regards,

[1]: https://github.com/ARM-software/bsa-acs/blob/v24.04_MEMTEST_REL1.0.0/val/sbsa/src/sbsa_execute_test.c#L543
[2]: https://github.com/ARM-software/bsa-acs/blob/v24.04_MEMTEST_REL1.0.0/val/common/src/acs_exerciser.c#L55
[3]: https://github.com/ARM-software/bsa-acs/blob/v24.04_MEMTEST_REL1.0.0/pal/uefi_acpi/common/src/pal_exerciser.c#L120

Sujana-M, 2024-12-24:
Hi @Azurithenkel ,

Thanks for raising this issue. The changes are present and binary is at https://github.com/Sujana-M/sbsa-acs/blob/Verification/image_to_test/Sbsa.efi. Can you please try this binary at your end?

Thanks,
ACS Team

Sujana-M, 2025-01-06:
Hi @Azurithenkel ,

Please let us know if your issue is resolved with the binary provided. 

Thanks,
ACs Team

vga-sipearl, 2025-01-09:
Hello, I am the one who opened the issue.

I took a look at the changes but I think the issue is not fixed.
The issue is somewhat cross-repositories since the issue has been found in the sbsa tests but the implementation of `pal_pcie_get_mcfg_ecam()` is in the bsa repository.

I think that the function `pal_pcie_get_mcfg_ecam()` should take the PCIe segment as an argument.
Without it, the function `pal_is_bdf_exerciser()` will always read in the ECAM of the first entry of the MCFG.
But a platform can have multiple non-contiguous ECAM spaces. In that case, an exerciser in the ECAM space of an entry of the MCFG that is not the first one will be reported as the first ECAM space.

Sujana-M, 2025-01-09:
Hi @vga-sipearl,

The changes in the commit are only in SBSA. But the binary mentioned has the respective changes. The binary at https://github.com/Sujana-M/sbsa-acs/blob/Verification/image_to_test/Sbsa.efi can be directly taken and run. 

Thanks,
ACS Team

vga-sipearl, 2025-01-09:
What are the exerciser Vendor/DeviceID in this build?

Sujana-M, 2025-01-10:
Hi @vga-sipearl,

The exerciser Vendor and Device ID in the build provided is 0xED0113B5. If the ID of the exerciser at your end is different, Please let us know, I will generate a binary with that for your verification.

Thanks,
ACS Team

vga-sipearl, 2025-01-10:
Could you publish the code somewhere or send it to us? We have some patches we need to apply for our setup.

Sujana-M, 2025-01-13:
Hi @vga-sipearl,

Please find the changes upstreamed on my fork.

SBSA: https://github.com/Sujana-M/sbsa-acs/tree/master and BSA: https://github.com/Sujana-M/bsa-acs/tree/main.

Thanks,
ACS Team

Sujana-M, 2025-01-22:
Hi @vga-sipearl and @Azurithenkel,

Please let us know if the issue is resolved with the upstreamed changes.

Thanks,
ACS Team

vga-sipearl, 2025-01-28:
Hello, sorry  for the delay. Trying to push the issue on my end.

vga-sipearl, 2025-02-05:
Hello,

Thank you for your patience. The Verification branches do indeed fix the issue on our end.

Thank you !

commit 550d3292 (2025-02-11): Fixes #521 Exerciser on multi-segments (#532)

- The UEFI PAL implementation to get the ECAM address
  was getting only from the segment 0 and not w.r.t
  the respective segments.
- Modified the PAL to check for the Segment and bus no
  of the BDF
closed 2025-02-11 by chetan-rathore