ARM-software/sbsa-acs #524: PMU SBSA Tests 1104 and 1105 need a way to change the start monitor

cdwilde, 2025-01-23:
With our system, we ran into a problem where some of the monitors were reserved and the defaults in these tests didn't work. I had to move the start monitor and had to edit the tests directly (as below). Is there a way we can get a PAL hook to specify a START_MONITOR?

**test_pool/pmu/operating_system/test_pmu004.c**

@@ -55,10 +56,10 @@ static uint32_t generate_inbound_traffic(uint32_t node_index, uint64_t base_addr
     val_memcpy(src_buf, dest_buf, size);
 
     /* Read the configured monitors for bandwidth values */
-    for (i = 0; i < NUM_PMU_MON ; i++) {
-        value[i] = val_pmu_read_count(node_index, i);
+    for (i = START_MONITOR; i < NUM_PMU_MON + START_MONITOR; i++) {
+        value[i - START_MONITOR] = val_pmu_read_count(node_index, i);
         val_print(ACS_PRINT_DEBUG, "\n       MON %d", i);
-        val_print(ACS_PRINT_DEBUG, " value = %x", value[i]);
+        val_print(ACS_PRINT_DEBUG, " value = %x", value[i - START_MONITOR]);
     }

     /*Free the buffers */
@@ -147,11 +148,11 @@ static void payload(void)
         }
 
         /* Configure PMEVTYPER to monitor Bandwidth value */
-        for (i = 0; i < NUM_PMU_MON; i++) {
-            status = val_pmu_configure_monitor(node_index, config_events[i], i);
+        for (i = START_MONITOR; i < NUM_PMU_MON + START_MONITOR; i++) {
+            status = val_pmu_configure_monitor(node_index, config_events[i - START_MONITOR], i);
             if (status) {
                 val_print(ACS_PRINT_ERR,
-                            "\n       Required PMU Event 0x%x not supported", config_events[i]);
+                            "\n       Required PMU Event 0x%x not supported", config_events[i - START_MONITOR]);
                 val_print(ACS_PRINT_ERR, " at node %d", node_index);
                 fail_cnt++;
                 break;

@@ -162,7 +163,7 @@ static void payload(void)
             continue;
 
         /* Enable specific monitors */
-        for (i = 0; i < NUM_PMU_MON ; i++)
+        for (i = START_MONITOR; i < NUM_PMU_MON + START_MONITOR ; i++)
             val_pmu_enable_monitor(node_index, i);
 
         /* Generate first memory traffic for 2 MB */
@@ -175,7 +176,7 @@ static void payload(void)
         }
 
         /* Reset the monitors */
-        for (i = 0; i < NUM_PMU_MON ; i++) {
+        for (i = START_MONITOR; i < NUM_PMU_MON + START_MONITOR; i++) {
             val_pmu_disable_monitor(node_index, i);
             val_pmu_enable_monitor(node_index, i);

**test_pool/pmu/operating_system/test_pmu004.c**

static void payload(void)
 {
@@ -99,8 +100,8 @@ static void payload(void)
         }
 
             /* Configure PMEVTYPER to monitor memory latency */
-        status1 = val_pmu_configure_monitor(node_index, PMU_EVENT_IB_OPEN_TXN, 0);
-        status2 = val_pmu_configure_monitor(node_index, PMU_EVENT_IB_TOTAL_TXN, 1);
+        status1 = val_pmu_configure_monitor(node_index, PMU_EVENT_IB_OPEN_TXN, START_MONITOR);
+        status2 = val_pmu_configure_monitor(node_index, PMU_EVENT_IB_TOTAL_TXN, START_MONITOR + 1);
         if (status1 || status2) {
             val_print(ACS_PRINT_ERR,
                       "\n       Required events are not supported at node %d", node_index);
@@ -132,15 +133,15 @@ static void payload(void)
 
 
         /* Enable the required counters to count programmed events */
-        val_pmu_enable_monitor(node_index, 0);
-        val_pmu_enable_monitor(node_index, 1);
+        val_pmu_enable_monitor(node_index, START_MONITOR);
+        val_pmu_enable_monitor(node_index, START_MONITOR + 1);
 
         /* Perform read/write and vary scale */
         val_memcpy(src_buf, dest_buf, BUFFER_SIZE);
 
         /* Read the monitors */
-        num_open_txn = val_pmu_read_count(node_index, 0);
-        num_total_txn = val_pmu_read_count(node_index, 1);
+        num_open_txn = val_pmu_read_count(node_index, START_MONITOR);
+        num_total_txn = val_pmu_read_count(node_index, START_MONITOR + 1);
 
         /*Free the buffers */
         val_mem_free_at_address((uint64_t)src_buf, BUFFER_SIZE);

chetan-rathore, 2025-02-10:
Hi @cdwilde,

The current PMU tests are written with coresight PMU architecture as reference (https://developer.arm.com/documentation/ihi0091/latest/)

The test follows the registers memory mapping as per that specification. Modifying the test to work with a different register memory mapping will require some investigations at our end. I will update you the timeline for same asap.

Thanks,
Chetan

cdwilde, 2025-02-10:
Are you saying that the PMU/MPAM tests are only applicable to the Coresight PMU Architecture at this time? Is there a way these tests can check for the architecture and skip if it is a different architecture? We got a lot of these working by modifying the ACS, but that isn't ideal. :)

chetan-rathore, 2025-02-11:
Hi @cdwilde,

The PMU and MPAM register offset and any bits field are part of VAL code as of now, so currently the test are applicable to coresight architecture.

We will check on providing a quick solution of skipping test if the PMU/MPAM are not as per coresight architecture.

Is it possible to share your PMU/MPAM architecture details/specification, it will help us to investigate and provide support for it.
We want to check if difference is between only the register offsets from PMU/MPAM base address, or there is also difference between bitfields.

Thanks,
Chetan

chetan-rathore, 2025-03-28:
Hi @cdwilde,

As discussed changes are done to skip the PMU test when the register layout is not same as the test expects.

Thanks,
Chetan

closed 2025-03-28 by chetan-rathore